PaulJenkin/PS4RaspberryPi #6: pi zero 2 w

pompeyjohnboy, 2021-12-23:
me and a lot of other members of psxhax have got pi zero 2 w model specifically to use with this exploit. If you could port these faster booting versions so that they can work with the zero 2 I'm sure it would be highly popular! Thank you

megaoctet, 2021-12-23:
Hello,
I join @pompeyjohnboy to support you and have a image for a pi zero 2w to test it for you if you want it.
Thanks for all you do for us.

PaulJenkin, 2021-12-23:
https://1fichier.com/?tup3vyaz7xucffn0g1az

Here we go

PaulJenkin, 2021-12-23:
https://1fichier.com/?4srzn1z2epvby5ij4k3j

Please test and let me know the results, waiting for it

megaoctet, 2021-12-23:
hello @PaulJenkin , 
it's perfect, it works great the first time for me.
Power cable connected to the usb port of the pi zero 2w
Only one cable for power supply and usb emulation 
Bravo!

PaulJenkin, 2021-12-23:
Cool thx for confirming

PaulJenkin, 2021-12-23:
Can you restart the PS4 and see how fast you can connect to the web page again?

pompeyjohnboy, 2021-12-23:
I wish I was having some luck. I've tried both of the test images and each time when I run the internet test it says connection failed and gives a DNS error. When I continue on to the browser and enter the 7.7.7.1 page I'm getting the exploit up and it does the first part correctly but then seems to hang on waiting for the PS4 pop on USB emulation, the bit where it should say GoldHEN, presumably as it's not happy about the DNS error. Obviously it's something silly I'm missing early on....

PaulJenkin, 2021-12-23:
@pompeyjohnboy 
Please try this one https://1fichier.com/?4srzn1z2epvby5ij4k3j

The first one had USB issue that is why it was removed
Also some wait time is expected to load the page , that is fine for now

megaoctet, 2021-12-23:
@PaulJenkin  @pompeyjohnboy  no worries for me with my console reboot. I had two insufficient memory and when I got the request to insert the usb key. I waited for 5 seconds and did ok and here is goldhen who arrived.
I have no connection problems with the first version posted above. I will test the second one but tomorrow. midnight here I'm going to sleep ;) . SSID/password : pizero / password

closed 2021-12-29 by PaulJenkin