PaulJenkin/PS4RaspberryPi #7: Updating JB files.

JasonDistler, 2021-12-23:
i have enabled SSH and am able to connect via wifi but cannot figure out where you are keeping the JB files for the server. I want to be able to keep this updated

PaulJenkin, 2021-12-23:
i understand the problem
the other way of doing it is update the data in SD card after its written (root user)

didn't add ssh as it was interfering with Apache/nginx , now i have isolated that issue, as i have already finalised the V5 version, will include the SSH Changes on V6

JasonDistler, 2021-12-23:
unless you can add samba like server to remote access JB folders

PaulJenkin, 2021-12-23:
Good idea will try do add in upcoming version

PaulJenkin, 2021-12-31:
Version 6 is uploaded which has the option to connect to local network

PaulJenkin, 2022-01-03:
Closing this as primary objective is reached

closed 2022-01-03 by PaulJenkin