PaulJenkin/PS4RaspberryPi #9: Feature request: WiFi Client Mode?

codefaux, 2021-12-29:
Simple request. I have a PS4, and a full homelab. I want to use Remote Play/etc.

I'd like to request a "Client" mode, wherein;
- RPiZero connects to designated (existing) WiFi network (new)
- RPiZero serves DNS, webserver as needed for PS4 exploitation (existing)
- RPiZero serves USB functionality (existing) 
- RPiZero uses static/DHCP to acquire its IP (new)
- RPiZero does *not* serve DHCP (new)
- PS4 acquires details via DHCP/manual static setup from user (at user's responsibility)

I know this targets a niche market, but it could be hugely useful for people who A) have a LAN, and B) wish to take advantage of the convenience this offers, without sacrificing Remote Play, FTP transfers, remote debugging over network, network package installers, etc. The PS4 could remain connected via ethernet, so long as it can reach WiFi clients like the RPiZero. It feels like this could be a relatively minor changeset, with huge potential.

If you're not interested in implementing this, perhaps you could share more details of your Buildroot configuration? I'd be willing to take a swing at implementing it myself, but I'm deeply unfamiliar with Buildroot. Thanks for your time!

PaulJenkin, 2021-12-29:
i have already started working on this, where the initial image will have PiZero Wifi netwrok.
Once connected there would an additional page in the JB in which user can select an existing WIFI network. once the WIFI network is selected the the Pi will join that network 
After this is done in order to connect to JB page user have to manually find the IP address of Pi and use it in PS4

JasonDistler, 2021-12-30:
perfect because i also like to use trainers :)

PaulJenkin, 2021-12-31:
Version 6 is uploaded which has the option to connect to local network

freak1234coder, 2021-12-31:
You have an error on line 162 in the RaspberryPi.php

change: 
<?php  (substr( $cmd, 0, 4 ) ===  '-rw-') : ?>
to:
<?php if (substr( $cmd, 0, 4 ) ===  '-rw-') : ?>

PaulJenkin, 2021-12-31:
@freak1234coder 
I need some help in publishing the host name to DNS Server . So that instead of Finding the IP from router page we can access the PI using hostname in PS4.
do you have expertise in this?

codefaux, 2021-12-31:
Not intending to step on toes, but;

If you mean you wish to use name resolution in the DNS and DHCP fields during network setup, the PS4 can't do that because it's a chicken and egg scenario. It needs that info in numeric form to be able to resolve names back to numeric form.

Assuming you mean in the browser window --

Every router is going to implement DNS overrides differently if at all, but most will support hostname resolution for the local network with no further configuration. Ie, with a local DNS (not using a public DNS for the PS4, not using router-supplied upstream DNS from your ISP, but the router itself serving DNS) it should 'just work' if you use the Raspberry PI's hostname. That won't redirect the User Manual to the exploit, or block updates, or anything else though.

That said, if you're using public DNS to block updates and so forth, you will never get local IP resolution by hostname or router override. They're mutually exclusive, which is why the Raspberry Pi method is superior - you can use local network options and still have the exploit hands-off. You don't want public DNS exploits with this method, as it also won't trigger the USB plugging.

The other problem with your question is, using the Raspberry PI method, the Raspberry Pi itself should BE the DNS server, aka every name that's asked for is returned as the PI for a destination. What you're running into is either that your PI doesn't have a predictable IP address, or you're allowing the router to give unmodified DNS servers to the PS4 over DHCP.

If your PI doesn't have a predictable IP address;

Try looking into adding a static DHCP mapping for the PI so that it always gets the same (currently unused unless it's the one it already has) IP address. This varies by router, of you need further help it's outside the scope of this thread; please seek further help externally. If the router came from your ISP, ask them about adding a static DHCP entry for your PS4. If they ask why it's too help with game stability. If the router is self-purchased, look to the manufacturer documentation for the same.

If you're letting the router give your PS4 DNS servers (aka "Easy" in Network Setup) you're letting it access the internet unfiltered, stop doing that because it isn't safe for the PS4 or your PSN account if you're logged in.

For proper configuration; Go to Network Setup.

Pick Wi-Fi or Wired, whichever you're using. Pick CUSTOM for "how do you want to set up the internet connection" AND NOT EASY. Pick your wifi network, passphrase, etc, if you're using wifi.

For IP Address I'm assuming you're using Automatic aka DHCP (or you wouldn't need this help) SO select Automatic unless you know better. Under DHCP Host Name, it's your call. If you give it a name here, your router should let you get at the PS4 via that hostname for things like FTP, Remote Play, payload sending, etc. If you don't, it auto assigns one. I forget what rules it uses..

DNS Settings MUST MUST MUST be set to the Raspberry PI's IP address (OR another manually configured Fake DNS server which points to the PI for exploitation and blocks Sony update servers/telemetry servers/etc. If you had that you wouldn't be here) -- set both entries to the Raspberry PI's IP address. If you set it to anything else the entire exploitation chain very likely will not work as intended.

Leave MTU at Automatic unless you know better.

Proxy Server to Do Not Use unless you know you need it.

That should cover the issue you're having, if not I apologize for taking your time. (PS this is the 'Users Responsibility' stuff I was talking about for the client mode proposal.)

Rocket200, 2021-12-31:
look in this he use dhcp
here my pi zero w get the ip and is online.
https://twitter.com/c0d3m4st4/status/1472376441832714243?s=21

PaulJenkin, 2021-12-31:
I am finding it hard to understand the above :)

My Requirement is 
after i joined the local Wifi Network i dont want to find the numeric IP from Routers page and use in PS4 browser
I like to use the hostname of PI in the PS4 browser. 
By default this functionality would have worked, Since we added the DNSMasq module, i think its screwing up the resolve conf.
I am trying to under stand what changes i have to do to make resolve conf work back

i am doing analysis on both side eventually will figure it

PaulJenkin, 2021-12-31:
@Rocket200 
https://twitter.com/c0d3m4st4/status/1472376441832714243?s=21
The above link is using full link stack, may be pi Zero 2 w is powerful enough to boot quick with the above build
but pi Zero will suffer

Rocket200, 2021-12-31:
or here some information 
https://github.com/ribamarsaraiva/roteiros/tree/main/raspberry_pi_zero_w

Rocket200, 2021-12-31:
> @Rocket200
> https://twitter.com/c0d3m4st4/status/1472376441832714243?s=21
> The above link is using full linux stack, may be pi Zero 2 w is powerful enough to boot quick with the above build
> but pi Zero will suffer

i use it with pi zero w and its work (i wait after reboot ps4 15 seconds

PaulJenkin, 2021-12-31:
@Rocket200
my aim to to boot the pi along with PS4 , so i have used embedded linux.
the above is using the Raspbian lite

If we go Raspbian route then custom chip of $2 or $3 is not possible.
as it cant run those

codefaux, 2021-12-31:
> I am finding it hard to understand the above :)
>
> My Requirement is after i joined the local Wifi Network i dont want to find the numeric IP from Routers page and use in PS4 browser I like to use the [...]

Okay - I replied assuming you were a user, didn't notice your were the author. Let's back up a step.

Most exploits of this style use a fake DNS server along with the exploit, so that *any* URL (specifically, the User Guide is the most convenient) is sent to the exploit server. I'm currently using an ESP8266 version which requires manual plugging of the USB drive, but provides the same fake DNS I'm mentioning.

My suggestion is to implement that Fake DNS server (there's two python versions on GitHub) on the Raspberry PI, and set the PS4's Network Settings to point at the Raspberry PI for DNS, thus any name given will grab the PI and exploit page, including just opening the User Manual.

codefaux, 2022-01-01:
To clarify now that I'm not on my phone in bed at 5am;

I've been trying to replicate the exploit stack locally on an existing LAN with opnSense as a router and a Linux stack running 25+ Docker containers and services (I have a lot of hobbies, lol) and from what I can tell, the "public DNS" exploits run hosted copies of more or less the same thing you'd find in the Al Azif repo.

To trigger those, you set your PS4's DNS server to their IP. Thus, any resolution query to "playstation.com" is sent -also- to them, where they forward the request to the exploit menu you normally see. By doing this, once the DNS is configured, you can simply load User Guide from the PS4's Settings page, and it will load the exploit instead of the User Guide.

Recently (9.00 and up) Sony has been using secure DNS requests, which some ISPs notice as being falsified and either correct or invalidate. I can dig up tweets discussing the issue if you're interested but that doesn't matter.

While trying to figure out how to implement it, I found an ESP firmware which serves the exploit and a fakeDNS server in the same way -- the User Guide is redirected to the exploit page for simplicity, saving the need to use the browser page, which obviously means you don't need to know an IP or even a URL to type.

The ESP firmwares simply direct -all- DNS inquiries to the module's IP address, and the httpd redirects 404 to the exploit index, so -any- browser request to -any- site is served the exploit -- thus also the User Guide.

The Al Azif self-host stack is more sophisticated and seems to load FakeDNS (https://github.com/Al-Azif/FakeDns) as a Pyhton module in a Python app, but it also operates as a stand-alone executable script in the repository it's forked from. 

In these cases, the key seems to be giving it a list of domains to redirect in its expected format, which is documented. Then, requests to PSN/Sony/etc are blocked (no updates, no PSN bans, no telemetry, no error reporting, etc) and the User Guide is redirected to the exploit index.

As an example of how FakeDNS is implemented in Al Azif's stack, one can find it in his repository but it's written as variables passed to a formatting function so it's not a clean read.

If you're not sure what to do with this information, I'd be absolutely pleased to help with implementation, but despite knowing enough to figure all of this out, I've not successfully used buildroot for anything, lol. If you could provide me with your buildroot environment (or just setup details) I would happily do the implementation work and return it.

PaulJenkin, 2022-01-01:
Please join the below discord Server
https://discord.gg/5XqUjx6X

d3structor84, 2022-01-02:
The Code won't handle Whitespaces in SSID.

Fixed it by changing:

Line 13 in /usr/html/schript.php:

From:
`$output = shell_exec('sudo /bin/sh /usr/html/Resources/Node/zrun.sh' .' '. $_POST['sid'] .' '. $_POST['pwd'] );`
To:
`$output = shell_exec('sudo /bin/sh /usr/html/Resources/Node/zrun.sh' .' "'. $_POST['sid'] .'" "'. $_POST['pwd']. '"');`

and Line 16 in /usr/html/Resources/Node/zrun.sh:

From:
`printf 'network={\n	ssid="'$1'"\n	psk="'$2'"\n	key_mgmt=WPA-PSK\n}' >> usr/html/Resources/Node/wpa_supplicant.conf`

To:
`printf "network={\n	ssid=\"$1\"\n	psk=\"$2\"\n	key_mgmt=WPA-PSK\n}" >> /usr/html/Resources/Node/wpa_supplicant.conf`

PaulJenkin, 2022-01-02:
@d3structor84 
To:
$output = shell_exec('sudo /bin/sh /usr/html/Resources/Node/zrun.sh' .' "'. $_POST['sid'] .'" "'. $_POST['pwd']. '"'); **- agreed**

To:
printf "network={\n ssid=\"$1\"\n psk=\"$2\"\n key_mgmt=WPA-PSK\n}"  -> this should have quotes

d3structor84, 2022-01-02:
> To: printf "network={\n ssid="$1"\n psk="$2"\n key_mgmt=WPA-PSK\n}" -> this should have quotes

unescaping the Quotes will result in a file structure like:
`ssid = asdf qwer`
but it should be:
`ssid = "asdf qwer"`

i dont know if its really necessary, but the resulting ssid should be quoted in wpa_supplicant.conf.

I'm really sure i tried to avoid the second change, but the results were as stated above.

Will try to change only the first part, to see if it works.

PaulJenkin, 2022-01-02:
i have changed the code to handle this in Php itself like
<?php 

$requestPayload = file_get_contents('php://input');
$jsonarr=json_decode($requestPayload, true);

if ($jsonarr['value'] == "RemoveUSB"){  
	$output = shell_exec('sudo /sbin/modprobe -r g_mass_storage');
}
if ($jsonarr['value'] == "LoadUSB"){ 
	$output = shell_exec('sudo /sbin/modprobe  g_mass_storage file=/usr/html/usb-drive.img stall=0 removable=1');
}
if ($jsonarr['value'] == "WifiRouter"){ 
	$output = shell_exec('sudo /bin/sh /usr/html/Resources/Router/zrun.sh');
}
if ($jsonarr['value'] == "Node"){ 
	$fp = fopen('Resources/Node/wpa_supplicant.conf', 'w');
	fwrite($fp,'network={'.PHP_EOL);
	fwrite($fp,'	ssid="'.$jsonarr['sid'].'"'.PHP_EOL);
	fwrite($fp,'	psk="'.$jsonarr['pwd'].'"'.PHP_EOL);
	fwrite($fp,'	key_mgmt=WPA-PSK'.PHP_EOL);
	fwrite($fp,'}');
	fclose($fp); 
	$output = shell_exec('/bin/sh /usr/html/Resources/Node/zrun.sh');
}
if ($jsonarr['value'] == "Reboot"){ 
	$output = shell_exec('sudo /sbin/reboot' );
}

if ($jsonarr['value'] == "updateDNSIP"){ 
	$output = shell_exec('sudo /bin/sh /usr/html/Resources/DNS/zrun.sh' .' '. $jsonarr['ipadd'] );
}

?>

PaulJenkin, 2022-01-02:
the SH file is not doing well when we have space hope above should work

PaulJenkin, 2022-01-03:
Closing this as the primary objective is reached

closed 2022-01-03 by PaulJenkin

codefaux, 2022-01-10:
Sorry I disappeared - had some personal life stuff explode on me unexpectedly, and I'm just getting back to things. If my help is needed I'll still gladly contribute, but it sounds like you've done superbly without it. Thanks guys, good luck, and link me to anyplace you need input, I'll gladly spend whatever time is needed if I can help.

PaulJenkin, 2022-01-10:
Sure No problem, thanks for your support