PaulJenkin/PS4RaspberryPi #10: GoldHEN needs updating to v2.0b2

DoctorWolfenstein, 2022-01-03:
The version of GoldGEN included in v6 has issues with the BinLoader, therefore App Dumper in v6 produces corrupted dumps (CE-80001-1 error on all dumps). A newer version of GoldHEN apparently fixes this (GoldHenV2**b2** link on Karo Host works fine)...

https://github.com/GoldHEN/GoldHEN/releases/tag/2.0b2

Also, is it possible to update the payload files manually via SSH/SFTP?

Thanks for your work on this project!

PaulJenkin, 2022-01-03:
The V6 itself allows SSH, username : root, password: password
However at present the GoldHen is converted as JS and being used.

I am planning to to change the JB to a simpler mechanism, if you have some better local host, let me know we can incorporate it.

PaulJenkin, 2022-01-07:
Implementation of V8 allows updating the goldHen bi file, so closing this issue

closed 2022-01-07 by PaulJenkin