PaulJenkin/PS4RaspberryPi #11: Does it work on banana pi zero board?

InfVRN, 2022-01-03:
Hello. In descriprion on aliexpress told about   compatibility this board, with rasberry pi zero images. Will this host work on this board? This boand is cheaper, than pi board. Thanks for answer and sorry for my english(

PaulJenkin, 2022-01-03:
by default it will not work as these are custom built of raspberry Pi, but using build root we can build one for banana pi well

InfVRN, 2022-01-04:
I can buy this board and test it with my ps4. Os on banana pi is here: https://wiki.banana-pi.org/Banana_Pi_BPI-M2_ZERO 
Can you look

PaulJenkin, 2022-01-04:
with Banana_Pi_BPI-M2_ZERO we can only do a blind development, i can develop but the chance of working is very less.
Pi Zero W should be a cheaper option i believe

InfVRN, 2022-01-04:
Thanks for answer. Yep, better buy pi ZeroW board.

closed 2022-01-04 by InfVRN