PaulJenkin/PS4RaspberryPi #14: ps4 offline trainers addition

JasonDistler, 2022-01-07:
V6 and V7 would not connect to my mesh network but V8 works perfect :)  if you can add the offline ps4 trainers then to me this will be the perfect AIO package

PaulJenkin, 2022-01-07:
not sure on how offline trainers work, do you have working example.
1. is it a web hosting?

JasonDistler, 2022-01-07:
Karo has esp8266 with offline trainers if that helps. https://mega.nz/folder/MHJCxLbJ#ilX7UeOlldGJpAzPaazr
https://karo218.ir/PS4TRAINER/index.html
https://github.com/KameleonReloaded/PS4OfflineTrainer
trainers all pulled from ps4trainer.com but i dont know how to pull them all for offline local use without caching a page but i would like 100% offline :)

JasonDistler, 2022-01-07:
Looks like Kameleon offline host is kept updated with the latest cheats so we can just ftp those over as needed. thanks for all your work

PaulJenkin, 2022-01-09:
Will add in next release

PaulJenkin, 2022-01-09:
https://github.com/PaulJenkin/PS4RaspberryPi/releases/download/Version9/PiZW-V9-Expermental1.zip
https://github.com/PaulJenkin/PS4RaspberryPi/releases/download/Version9/PiZ2W.-.Expermimental1.zip

These have trainers , let me know if its good

JasonDistler, 2022-01-09:
Works great thanks
things i noticed:
Initial wifi setup (connect to pizero 7.7.7.1 page) get weird numbers and characters in SSID scan and my mesh network shows but after adding password and rebooting the pi never connects (probably just need to be patient and give it time on initial boot. refresh page and fixed. 
trainers work great :)
just a reminder for the user to run webrte900/ webrte payload first for fw9.00 or older FW
also may need to set the trainer page to search "localhost" in the IP field and if game is running the trainer should attach. the trainer menu will tell you when attached

thanks again this is awesome.

PaulJenkin, 2022-01-09:
Can you elaborate on "get weird numbers and characters in SSID scan and my mesh network shows but after adding password and rebooting the pi never connects" , probably screenshots?

JasonDistler, 2022-01-11:
just tried multiple more times with clean image installed and no issues

PaulJenkin, 2022-01-12:
Closing the issue

closed 2022-01-12 by PaulJenkin