PaulJenkin/PS4RaspberryPi #24: Not pop up the USB drive.

langerst, 2022-01-22:
it's not pop up the USB drive after notice came say wating.....blabla....anything more I should do?.......or any place setting I can check....?

langerst, 2022-01-22:
V10 RsZW trying many times.....

PaulJenkin, 2022-01-22:
What Device you using?

1. If Pi Zero W or Pi Zero 2 W, the micro USB cable has to be in USB Slot, All not all USB wires support Data transfer, some wires are just for power, Try with different wires

langerst, 2022-01-23:
thanks for reply, Pi ZeroW using the USB slot, the cable tested with gamepad working with data. Im thinking about the port broken?......or some setting need to be reset on Pi bios.....

langerst, 2022-01-24:
Starting php-fpm
done
etc/init.d/rcS: line 23: /etc/init.d/S50avahi-daemon: Permission denied
Starting nginx.
Starting sshd: OK
Starting dnsmasq: OK
Starting hostapd: Configuration file: retc/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:b:8c:ba:34 and ssid "PiZero"
wlan0: interface state UNINITIALIZED-›ENABLED
wlan0: AP-ENABLED
OK
Starting pyFTPServer OK
nohup: appending output to nohup. out
/etc/init.d/rcS: line 23: etc/init.d/S99IdelTimer: Permission denied
Welcome to PiZero
pizero login:

langerst, 2022-01-24:
`Starting php-fpm
done
etc/init.d/rcS: line 23: /etc/init.d/S50avahi-daemon: Permission denied
Starting nginx.
Starting sshd: OK
Starting dnsmasq: OK
Starting hostapd: Configuration file: retc/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:b:8c:ba:34 and ssid "PiZero"
wlan0: interface state UNINITIALIZED-›ENABLED
wlan0: AP-ENABLED
OK
Starting pyFTPServer OK
nohup: appending output to nohup. out
/etc/init.d/rcS: line 23: etc/init.d/S99IdelTimer: Permission denied
Welcome to PiZero
pizero login:`

closed 2022-01-24 by langerst