PaulJenkin/PS4RaspberryPi #26: QOL Improvement suggestions

AlfieCH, 2022-01-24:
Hi. First, let me thank you for this project. It work great.

I have two suggestions for QOL improvement :

1. 2nd click ("Jailbreak done") is not useful.
I removed it and it work without problem (I just removed alert at line 647 in kexploit.js).

2. I modified index.php so when menu is opened, "GoldHEN" button is in the middle of the screen, right under default cursor position.
For my usage, I created a simpler index.php page with only 4 buttons : Threshold, Raspberry, GoldHEN and "More options" (open original index.php I renamed)

With those 2 modifications, you just need to open guide, click, wait and click. Very easy :)

PaulJenkin, 2022-01-24:
Can you Share you index.php?

1. 2nd click ("Jailbreak done") is not useful. - this was introduced only to avoid the flickering in ps4 slim when comming back from JB to webpage

2. I modified index.php , send it to me will integrate into the code

ribamarsaraiva, 2022-01-24:
This project is amazing, just if possible you can make better using https://github.com/stooged/PS4-Server-900u, @karo_sharifi implemented a esp8266 host using that idea, if you want see here is the bin file https://t.co/9buavulD5x

AlfieCH, 2022-01-24:
Here is my index.php.
Note :
I left all original code, so it's easier for me to maintain.
Note 2 :
This file fits my needs (only using GoldHEN. Someone using payload or other options may not be happy to have more clicks to do.
[index.zip](https://github.com/PaulJenkin/PS4RaspberryPi/files/7928212/index.zip)

AlfieCH, 2022-01-24:
[index.zip](https://github.com/PaulJenkin/PS4RaspberryPi/files/7929294/index.zip)

PaulJenkin, 2022-01-26:
Keeping this handy will it try and incorporate in next release

closed 2022-01-26 by PaulJenkin