PaulJenkin/PS4RaspberryPi #27: System Volume Information visible in folder list

AlfieCH, 2022-01-24:
If you plug SD card on Windows, Windows automaticaly create a (hidden) folder named "System Volume Information".
This folder is visible in menu (bottom button) after "GoldHEN", "Linux" and "Mod Menu". 
It should be excluded.

PaulJenkin, 2022-01-24:
Yup
As I was a Linux user didn't notice it
But I have fixed it with below logic
Display folder only when bin count gearter than 1
Will be available in v11

AlfieCH, 2022-01-24:
Here is my index.php.
Note : 
I left all original code, so it's easier for me to maintain.
Note 2 :
This file fits my needs (only using GoldHEN. Someone using payload or other options may not be happy to have more clicks to do.
[index.zip](https://github.com/PaulJenkin/PS4RaspberryPi/files/7928197/index.zip)

PaulJenkin, 2022-01-26:
Should fixed in V11, check it release page

closed 2022-01-26 by PaulJenkin