PaulJenkin/PS4RaspberryPi #30: Usb Emulation probleme? PiZ2W v11

megaoctet, 2022-01-26:
Hello,
Bravo for all this work.
Just to tell you that with the last version on pi zero 2 W PiZ2W-v11-Build1.zip . the usb exploit does not launch auto anymore, I never have the pop up that indicates that a usb key has been inserted. It worked perfectly on the 10 build6.zip
Maybe I'm the only one ? 
thanks

dem0nb0y, 2022-01-26:
I'm having that same issue too v10 worked on Auto JB but v11 is not.

PaulJenkin, 2022-01-26:
Can you check?
https://github.com/PaulJenkin/PS4RaspberryPi/releases/download/Version11/PiZ2W-v11-Build2.zip

megaoctet, 2022-01-26:
Yes of course 
I test directly on my console 
No change the pop up of the usb key does not appear.
and if I do "ok" I get a message saying "failed to trigger the exploit, ...)

dem0nb0y, 2022-01-26:
V10 is the only one that is working so put v10 back on until it's fixed.

PaulJenkin, 2022-01-26:
@dem0nb0y Ok Gangster :)
V10 is already available in release, so anyone can download it, Rebuilding Pi Zero 2 W from scratch

dem0nb0y, 2022-01-26:
> @dem0nb0y Ok Gangster :) V10 is already available in release, so anyone can download it, Rebuilding Pi Zero 2 W from scratch

What i meant was Zero 2W V10 is the only one that works. In V11 Auto JB is broken and so is the Exploit and GoldHEN.

PaulJenkin, 2022-01-26:
No issue i understood , Just kidding , thanks for testing

dem0nb0y, 2022-01-26:
> No issue i understood , Just kidding , thanks for testing

You're welcome, Also thanks for this awesome automated jb i love it is there anyway of getting it to to press the ok buttons automatically too? Like say you restart the PS4 and leave it unattended. And another feature like adding PKG's via FTP client to the memory card once this v11 is fixed?

megaoctet, 2022-01-26:
Thank you for all that you do 
All the way to v10 is already amazing 
v11 will be perfect for sure

PaulJenkin, 2022-01-26:
@dem0nb0y , FTP to Pi will be dead slow for sure, we still want to make that happen if interested.
for removing SD card and inserting in PC should be fine

dem0nb0y, 2022-01-26:
> @dem0nb0y , FTP to Pi will be dead slow for sure, we still want to make that happen if interested. for removing SD card and inserting in PC should be fine

Saves having to remove the SD card since i have my Pi Zero2W in a case it's a nightmare having to keep remove it to take the sd card out. Future updates would be a great addition so updating via the pi if that's possible.

PaulJenkin, 2022-01-26:
https://drive.google.com/file/d/1p3LWj9bJzeqXBbk64D7kLrLIkCs9IeUb/view?usp=sharing

PaulJenkin, 2022-01-26:
will be publishing the root Password with this release
so SFTP access to all folders no should not be an issue

megaoctet, 2022-01-26:
> https://drive.google.com/file/d/1p3LWj9bJzeqXBbk64D7kLrLIkCs9IeUb/view?usp=sharing

Non pas de changement pour moi
La led clignote bien une fois comme si l'action allait se lancer mais pas de pop up  de cle usb

PaulJenkin, 2022-01-26:
Clean Build is also not working, Will try adding patches on top of V11 and see if its picking it up

PaulJenkin, 2022-01-26:
check this
https://github.com/PaulJenkin/PS4RaspberryPi/releases/download/Version11/PiZ2W-v11-reBuild1.zip

megaoctet, 2022-01-26:
> check this https://github.com/PaulJenkin/PS4RaspberryPi/releases/download/Version11/PiZ2W-v11-reBuild1.zip

It's  OK !!  I haven't tried the auto mode but in normal manual launch it's perfect  GG

closed 2022-01-26 by PaulJenkin