PaulJenkin/PS4RaspberryPi #32: OG Pi Zero support

shycoderX, 2022-01-26:
Would it be possible to add support for the non W Pi Zero?

PaulJenkin, 2022-01-26:
Does it have a way to connect using Ethernet?

shycoderX, 2022-01-27:
> Does it have a way to connect using Ethernet?

Yes using an USB to Ethernet dongle. You can also use a Wi-Fi dongle. But you need to connect a hub to the Pi then ofcourse so you can connect a usb port to the PlayStation too.

EliasFogarty, 2022-01-30:
If you can SSH into the Pi try the command 'ifconfig' to find your adapter name (ex: wlan0)

shycoderX, 2022-02-01:
> If you can SSH into the Pi try the command 'ifconfig' to find your adapter name (ex: wlan0)

AFAIK you need your PI connected to the internet to access SSH. I tried connecting to it via serial using an adapter (which works on RPi OS) but to no avail.

shycoderX, 2022-02-01:
I think some drivers for USB Hubs and WiFi adapters are just missing which are available on RPi OS. As it detects my Hub and my WiFi adapter (as wlan0).