PaulJenkin/PS4RaspberryPi #33: SFTP Root and Shutdown

dem0nb0y, 2022-01-27:
Thanks again for the fix Paul it's been working flawless, I have a question regarding the root password to SFTP in to the Zero 2W to gain access to all folders including the PKG Files to add via FTP do you know what it is please? And has for shutting down the Pi does it need to be shut down before you shut down the PS4 to avoid FS Corruption? It wasn't mentioned in the instructions.

PaulJenkin, 2022-01-29:
Adding Power shutdown before PS4 Turn off cannot be achive, Until unless we are able to pass something from PS4 to Rpi during shutdown.
Also FS corruption wont happen, until unless PS4 is the only client of PI

dem0nb0y, 2022-01-29:
> Adding Power shutdown before PS4 Turn off cannot be achive, Until unless we are able to pass something from PS4 to Rpi during shutdown. Also FS corruption wont happen, until unless PS4 is the only client of PI

Thanks Paul, when you say "until unless PS4 is the only client of Pi" do you mean FS Corruption will not happen if its just the PS4 it's connected to so it's safe to restart or shut down the PS4 While Pi is still ON? Any update on your project? managed to get root working on v10 via sftp but it doesn't work on v11.

PaulJenkin, 2022-01-29:
Yes If RPI is just connected to PS4 and PS4 turnoffs FS coruption will not happen,
if you want to PI to turn off automatically check on idle timer as well

dem0nb0y, 2022-01-29:
> Yes If RPI is just connected to PS4 and PS4 turnoffs FS coruption will not happen, if you want to PI to turn off automatically check on idle timer as well

Thank you Paul.

closed 2022-01-29 by dem0nb0y