PaulJenkin/PS4RaspberryPi #34: Auto JB issue (Z2W V11)

Pharaoh2k, 2022-01-28:
When trying to auo JB, it always reaches either Messages or Party and enters there instead of going to Settings.
I believe the reasons is that I got a lot of games, especially on the extended storage and the PS4 scans the ext storage when logging in, which slows down the whole XMB. Even if I manually try to move with the controller right after logging in, while the scanning is running - some steps are skipped./not registered

BRK65, 2022-02-02:
I have this same issue on a PI Zero W board. IT worked correctly the first couple of times but it usually ends up going into Notifications on mine and ends up clicking the top one.  It will jailbreak the console after this if I manually run it but the Auto JB would be nice.
Paul,
It's like Pharoah2K said above, it seems like it starts trying too soon. Is there a timeout variable we could set to give the PI a bit more time before it starts attempting to JB? My console takes a bit before it is properly responsive to the virtual keyboard.
Thanks for such an awesome tool btw!

BRK65, 2022-02-03:
I was able to solve my issue by changing the boot_delay=0 to boot_delay=40 in the config.txt file in the boot partition on the SD card. Played around with different timings and found that one worked best.
Pharaoh2k
You may want to try the same. I started with 60 and worked my way down.

Pharaoh2k, 2022-02-05:
@BRK65, nice workaround. Thanks! :)
Hopefully, a permanent solution will be available soon.

BRK65, 2022-02-06:
@Pharaoh2k  No problem! :)
I've been tweaking a lot of things in it on mine. It's definitely a cool tool!
I don't think there is another way to do it if you have a ton of games on an external drive. The PS4 is just really slow to start when doing all that scanning before full startup.