PaulJenkin/PS4RaspberryPi #37: PZW Failed trigger the Exploid

PittStone, 2022-01-29:
I have a fresh install of PiZW-v11-reBuild2 and after setup DNS and go to the Page for Goldhen it shows the Notification for USB Emulation starting. Than i waiting over and over Minutes and nothing happens. If i klick O.k. a Message shows "Failed Trigger the Exploid..."

PaulJenkin, 2022-01-29:
Have you tired Rebuild 1 is it working for you?
If that is also not working i would do the below
1. Check if micro USB is plugged in the USB slot of PI
2. Change the cable which you are using as there are some cable which are only for power and not for data

PittStone, 2022-01-29:
If i change to load "goldhen_2.0b_900.bin" it fails, too. Only i can activate "goldhen_2.0b2_900.bin".

PittStone, 2022-01-29:
Loading other goldhen works if i close the Userbook and open it again.  But the same Error happens with Rebuild 1! Its failed to trigger the Exploid.  I have also changed to a external Power adaptor without plug in usb to PS4.

PittStone, 2022-01-29:
After i press o.k.  A System Message appears "The free system memory is insufficient".  My Free System Memory is 89GB.

PittStone, 2022-01-29:
After other two Tests, another USB Cable was the right for Data-transfer pi to ps4. Thats why nothing Notification happens.

closed 2022-01-29 by PaulJenkin