PaulJenkin/PS4RaspberryPi #42: Goldhen needs updating to 2.1

notimp, 2022-02-05:
Internal update function doesnt seem to work either in v11_2. I dropped the new bin In the folder suggested in the readme, and all I get in the web interface is one folder (all other bin related options vanished), and inside the folder, the bin file.

No "update Goldhen" to be found anywhere.

Pasha1402, 2022-02-05:
It is actually green button in right-bottom corner called GoldHen, which allows to select version of GoldHen to use. Though after adding and activating it behaves odd for me: normal sequence followed by "No payload available" pop up and nothing happening.

notimp, 2022-02-06:
I thought that button was only there to select the .bin manually. :)

Reason, to have the .bin show up in that menu (folder really) you drop it into the Goldhen directory on the user accessible partition.

To "update" Goldhen according to the readme, you have to drop it into a folder in the root partition (using a linux/unix file system, so windows users wont see it), if you do that - all other options on the (html) main screen vanish - and you only see the goldhen folder, but nothing else happens.

To "update" Goldhen (via a script, not just by selecting it) is necessary, because according to the readme, it needs to get converted to a .js file, which the option you meniton doesnt do.

Also "selecting it manually" isnt something I'm interested in, as I use this project mainly for the auto JB function, so implementation with that feature is important to me.

Long story short - either there is a dependency on filename that isnt mentioned in the readme, or the update function broke somewhere along the way (on newer releases), either way - this needs someone who knows the internals of this projects to look at whats happening currently, and maybe provide a fix - or a newer release with Golhen already integrated -- but that would mean more work for the maintainer, with new releases with every Goldhen update.

What you are mentioning is not the "solution" to this issue (updating the main Goldhen payload doesnt seem to work), not even conceptually. ;)

Jonny0r, 2022-02-06:
I know this is not helpful, but will there be an 12th release with GoldHen 2.1? or do we have to get it working somehow by our self?

BRK65, 2022-02-06:
The easiest way to get the access you want to the files you can't see in Windows is to add SSH connectivity to the Pi by just creating an empty file called ssh in the pBoot partition on the SD card from your PC. Make sure you have file extensions turned on on your PC as it just needs to be a blank file called ssh without any file extension(no ssh.txt or anything like that). Once you have done that and booted the Pi back up, you can connect to the Pi in most FTP clients by changing the protocol to SSH over SFTP or just SFTP in some clients. The user name will be root and the pass <?>123   on port 22 (It will ask you to save and trust the key when connecting)
The file can be placed in /usr/html    ...    The old bin should be deleted from there. Be careful what you change or you could end up having to re-image your Pi's SD card.

Pasha1402, 2022-02-06:
ssh already enabled in this image by default. there is no reason to delete old file if new one has different name. wonder which php or js tells what bin actually to pick?

Pasha1402, 2022-02-06:
I did check what GoldHen activation script does: it is actually makes changes mentioned by BRK65 above: file in /usr/html named goldhen_2.0b2_900.bin is being replaced by goldhen2.1_900.bin. Nothing works after it... assuming renaming GoldHen 2.1 into 2.0b2 without using activation button may be workaround, but not the solution

notimp, 2022-02-07:
Thank you @Pasha1402 .

Pharaoh2k, 2022-02-07:
Works fine for me, but I did it in a different way than suggested.
I logged in to the pi FTP server, entered `/GoldHen` folder, uploaded the `goldhen_2.1_900.bin` to there.
Then back to the XMB opened the pi HTTP URL, clicked the GoldHen folder, selected the new version and got a prompt that from now it will be the one loaded.
That's all that was needed.

BRK65, 2022-02-07:
@Pharaoh2k I didn't realize that was an option in v11. I had tried that in v10 and just got an error

Pasha1402, 2022-02-07:
Found some time to re-flash image from scratch. Now it works for whatever reason it was: new GoldHen version placed in GoldHen folder and activated via button with the same name, copies it properly into web root and whole voodoo thing works as expected. Make sure you are using build**2**. Think this issue may be closed now.

h4x0rm1k3, 2022-02-08:
It does update fine but I ended up going the other route because putting the bin file in the folder didn't work at all. So, I connected my PC to the pi using the PiZero wifi connection and then in Filezilla changed the protocol to FTP and put 7.7.7.1 in the address box and 2121 in the port box, then I used username - user and password - fusand . It then opened up in to the Pi's folder views where I could see a Goldhen folder and inside there is the 2 old Goldhens, I deleted the really old one and replaced it with the new version, then I went through the guide on PS4 to get to the Webhost server, clicked the green Goldhen around bottom right of the selection boxes and in the new window I could then see the new version of Goldhen which I ran and it updated itself and now the main Goldhen button on the main screen will install Goldhen v2.1 9.,00 instead of 2.0.2b or whatever it was before.

notimp, 2022-02-08:
Thank you for figuring it out. :) e/o

Pharaoh2k, 2022-02-17:
@notimp I think it's better to close this issue...

notimp, 2022-02-25:
Done (read it fairly late).

closed 2022-02-25 by notimp