PaulJenkin/PS4RaspberryPi #44: Source code for image?

MrMEEE, 2022-02-17:
Hi

Is the source code / build pipeline available somewhere?

I have some changes/enhancements I would like to try to implement and upstream?

neclegens, 2022-02-20:
You can write to some media (i.e. an sd card) and read the contents from there which should let you update GoldHen etc.

kicolassilva, 2022-03-16:
What software can I use to edit the original file in order to update GoldHen and make improvements?

CrazeeGhost, 2023-01-04:
@PaulJenkin would you be willing to open source this? I'd love to be able to run the host and exploit on a clean Raspberry Pi OS and run additional services on the same box.

SpectreDev-007, 2023-07-18:
> @PaulJenkin would you be willing to open source this? I'd love to be able to run the host and exploit on a clean Raspberry Pi OS and run additional services on the same box.

https://github.com/SpectreDev-007/PS4RaspberryPi 

can you share your build once done

CrazeeGhost, 2023-07-18:
> > @PaulJenkin would you be willing to open source this? I'd love to be able to run the host and exploit on a clean Raspberry Pi OS and run additional services on the same box.
> 
> https://github.com/SpectreDev-007/PS4RaspberryPi
> 
> can you share your build once done

I am actually working on it. I might have something to share in a couple of weeks

CrazeeGhost, 2023-07-20:
> Hi
> 
> Is the source code / build pipeline available somewhere?
> 
> I have some changes/enhancements I would like to try to implement and upstream?

> What software can I use to edit the original file in order to update GoldHen and make improvements?

Check out https://github.com/CrazeeGhost/PS4JbEmu