PaulJenkin/PS4RaspberryPi #47: Fail to boot on  PiZW 1.1 with PiZW-v11-reBuild2

dvlowe, 2022-02-25:
I have a Raspberry Pi Zero W (version 1.1). When I load the PiZW-v11-reBuild2 (or PiZW-v11-reBuild1) and power it up I get a green led that blinks twice then goes off and nothing happens.  

I am trying to get my hands on a  Raspberry Pi Zero 2 W for MSRP but that seams impossible at this time.

Thanks in advance for any assistance.

PaulJenkin, 2022-06-20:
Please check your board there is Raspberry Pi Zero board without wifi, and its not compatible

Pumpytums, 2022-07-27:
Oddly I have exactly the same issue. Tried it on 2 different Zero W ver 1.1 two blinks and nothing.

I tried the image for the Zero 2 W on my Zero 2 W using same card worked perfectly.

weirdcity, 2022-08-29:
Same, did you ever get it to work? 

>

Pumpytums, 2022-08-29:
Sadly no. Works great on Zero 2 W not on Zero 1.1 W. I tried it on two different boards.

FrustratingKill, 2022-12-26:
Same here. Stopped working today for no apparent reason and reflashing on a new SD card didnt help either. Tried with 2 different boards

jmp49152, 2023-01-30:
I read the thread here after having issues with a Zero W ver 1.1, and dug into the issue.

1) two blinks and then nothing is normal behavior for a Zero W 1.1 and this image, it blinks to say its started to bootstrap, then turns off the led, but its still running. If you hook a monitor up with a hdmi adapter to mini, you can see the boot messages as it does this.

2) I came here because I had problems with a Zero W 1.1 that wouldn't boot, then while diagnosing the issue with a keyboard and monitor to see if I could fix it for our variant, I re-flashed the PiZW-v11-reBuild2.img image to a different sd card (using dd under linux...) and suddenly it started to boot fine.

3) Bonus thing, if you have one about, the tiny cable that comes with 8bitdo interfaces is great for connecting the zero w up to the ps4.

So to confirm, I've got a Zero W 1.1 variant working fine with the image in the releases.