PaulJenkin/PS4RaspberryPi #50: Payload is empty error

P4RI4H, 2022-03-30:
I tried updating the GoldHEN bin both using ftp and copying directly to the sd card, and the update seems to work fine, however when I run the GoldHen exploit, it goes through the motions until I press X. It returns "Payload is Empty," or a similar message. 

I've tried with 2.1.2 and 2.2 and 2.2.2 from the GoldHEN github. I'm using the FW 9.00 bin as well. I'm not sure if I'm missing a step or setting. Thanks!

notimp, 2022-04-09:
After enabling it. And maybe rebooting the PS4 and the Pi. If you still get the error, then simply click on Goldhen up top once, and it should work from now on.

I guess how it works is as follows. Setting the payload only changes the "path" to the new named payloadfile - but as js (I think its js, could be misremembering).

Basically, the payload isnt executed as a .bin file, but first converted ito .js (I think), so when you choose a new Goldhen FW, the converted version of the payload does not exist yet, and you get the "not found" error.

When you then click on Goldhen (up top) once, the file gets created, and from then on everything works fine. Cant remember if you _need_ to reboot between selecting the goldhen bin file, and pressing goldhen up top. Probably not.

P4RI4H, 2022-04-12:
It's when I choose GoldHen at the top that I get this error. The one to choose payload wors fine and the Pi reboots itself (I believe, I haven't tried it in a bit). When I try executing the GoldHen it exploits but give this error. Rebooting never fixes it for me and re-selecting 2.0b never fixed it either. I had to rewrite the image file to the micro sd.

closed 2022-04-12 by P4RI4H