PaulJenkin/PS4RaspberryPi #62: Update to utilise PSFree exploit

haydg, 2024-01-30:
Is it possible to update this setup with the newer and faster PSFree exploit?

SpectreDev1, 2024-02-01:
working on it here https://github.com/SpectreDev1/PS4RaspberryPi

kylaris, 2024-02-07:
> working on it here https://github.com/SpectreDev1/PS4RaspberryPi

I tried building the image using your files but I am getting errors with the fakeroot.lo and I do not know how to fix this( First time using makefile)
Can you release the built image? thank you.

SpectreDev1, 2024-02-07:
Ignore that try this one

https://github.com/SpectreDev1/PSFree-RPI

(Still WIP)

SpectreDev1, 2024-02-13:
https://github.com/aladie/PS4RaspberryPi

someone else has beaten me to it, works nicely as i tested

ignore me

PaulJenkin, 2024-02-14:
Cool some one reversed my work

Silent-Remux, 2024-02-18:
> https://github.com/aladie/PS4RaspberryPi
> 
> someone else has beaten me to it, works nicely as i tested
> 
> ignore me

@aladie When updating goldhen to 2.4b, I get "no payload data" after the jailbreak done please click ok.

manishdixit1978, 2024-05-22:
why my USB emulation not happening?