PaulJenkin/PS4RaspberryPi #64: increase memory size

Buledo1, 2025-02-18:
hi,

the new goldhen payload seems to cause issues when loading. goldhen support mentioned - alot of the device code is based on old bin server code and doesnt alocate enough memory as latest beta is bigger than 0x41000 and this is why it most likely crashes. could you look into this please as your rasperry pi solution is simply easy to use and update.

thanks.

kekkodance, 2025-04-29:
this is true

tejesh83, 2025-08-09:
Thanks for positing @kekkodance and @Buledo1!

Any one had luck getting this to work with the latest goldhen 2.4b18.5?

In the original Paul Jenkin release, I tried replacing with 0x60 but it still crashes when trying to load this goldhen version.   I also tried this fork, https://github.com/aladie/PS4RaspberryPi, and also tried replacing it with 0x60, but no luck.  

@kekkodance do you mind sharing a snippet from the mjs file showing what changes you made?  I'm wondering if I edited the wrong references to 0x41, since it pops up a few times in the script.  Thx!